arduino/Arduino #7423: Arduino IDE portable mode on macOS is undocumented

per1234, 2018-04-06:
From https://www.arduino.cc/en/Guide/PortableIDE:
>This document explains how to make a portable installation of the Arduino Software (IDE) on Windows and Linux machines.

The lack of mention of macOS/OS X in the above sentence gives the impression that portable mode may not be supported on that OS. The IDE installation folder structure is different on macOS, which makes the folder structure shown on that page not very instructive for macOS users.

----
Suggested changes to that page:

Change:
>This document explains how to make a portable installation of the Arduino Software (IDE) on Windows and Linux machines

To:
>This document explains how to make a portable installation of the Arduino Software (IDE).

----
Change:
>The tree structure should be as this one; the

To:
>On Windows or Linux, the tree structure should be as shown above. On macOS the portable folder should instead be created under Arduino.app/Contents/Java/. The

----
If you want to add a macOS-specific screenshot of the correct folder structure that would also be helpful.

----
Originally reported at http://forum.arduino.cc/index.php?topic=539551

carlosperate, 2018-04-06:
👍 +1 On documenting this.

I've also gone through the trial and error process of creating a "portable" folder inside different directories in the app bundle. However, once I add it to the `Arduino.app/Contents/Java/` directory I encounter the following error, which I wasn't able to solve:

[image]

If I try to open it with sudo I get:

```
> sudo open Arduino.app
LSOpenURLsWithRole() failed with error -10810 for the file /Users/carlosperate/Downloads/Arduino.app.
```

SimonePDA, 2018-04-11:
I spoke with the Arduino Software IDE developers and the Portable version is not officially supported on Mac OS because of the package signature and the OS reactions to any change in the package content. The portable installation should be a simple procedure and unfortunately on Mac this is not the case. I have amended the Portable IDE page to document this and I also linked the forum thread so that the community efforts are available to the other users that would like to experiment.

closed 2018-04-11 by SimonePDA

carlosperate, 2018-04-13:
It's unfortunate that it can't be supported in macOS, but it's understandable.
Thanks for adding the note to the docs!