arduino/Arduino #7435: [Library Manager] Please add library for Grove - Temp&Humi&Barometer Sensor(BME280)

lanselambor, 2018-04-09:
release: https://github.com/Seeed-Studio/Grove_BME280/releases/tag/v1.0.0
github: https://github.com/Seeed-Studio/Grove_BME280

cmaglie, 2018-04-09:
`Grove - Temp&Humi&Barometer Sensor(BME280)`

same here, no `&` nor `()` in the name.

lanselambor, 2018-04-10:
Fixed that, thank you! 😃

per1234, 2018-04-10:
The name still contains the disallowed character `&`:
https://github.com/Seeed-Studio/Grove_BME280/blob/1.0.1/library.properties#L1

lanselambor, 2018-04-10:
Please check  it again.

lanselambor, 2018-04-16:
@per1234 please add this library to Library Manger,  thanks a lot.

per1234, 2018-04-16:
I don't have any power to add libraries to the index; I just try to help out a bit with administration of the issue tracker. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. I removed the `Waiting for feedback` tag after you resolved the issue so that means this will be processed on the next run through the requests. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2018-04-16:
done

closed 2018-04-16 by cmaglie