arduino/Arduino #7447: Bug with auto-selecting the right board and port when switching between two scripts

mirkosaiu, 2018-04-09:
As you can see in the image below I have two scripts opened and I'm working on them contemporarily. They belong to two different boards respectively. When I switch between them the IDE tries to update the port and the board but without success. In fact, in the Tools panel the preview shows something different from what is actually selected.

[image]

facchinm, 2018-04-10:
Hi @mirkosaiu ,
this looks like the bug is only graphical; can you confirm this? If you press compile on either window does it retain the correct target?

mirkosaiu, 2018-04-11:
What is selected in the of the available boards and ports is what is actually selected for real. But I think that the feature that automatically switches between the respective boards and ports of the script would be great!

facchinm, 2018-04-11:
The feature you are referring to is already merged into the Beta channel (from https://github.com/arduino/Arduino/commit/6112d0455ea66a3318739f2bd5de11ea504dec00) but it only works if the serial port has an unique vid/pid combination (unlike most ESP boards).
Feel free to test it (https://www.arduino.cc/en/Main/Software#beta) and report your impressions :wink:

mirkosaiu, 2018-04-11:
Ok, thanks :)