arduino/Arduino #7455: [Library Manager] Library addition request - Unified A/D multiplexer interface

stechio, 2018-04-11:
Apparently, the Arduino library collection lacks a **unified interface for controlling analog/digital multiplexers**, so I decided to develop an elegant solution based on Nick Lamprianidis' MUX74HC4067 project (full credits included in the library). Relevant features:
* generic Mux + specialized implementations (4051 and 4067 ICs, at the moment)
* encapsulated R/W A/D operations
* multiple signal pin switching

Please consider adding this library to the Arduino library collection:

github: https://github.com/stechio/arduino-ad-mux-lib
release: https://github.com/stechio/arduino-ad-mux-lib/releases/tag/v1.0.0

Thank you

per1234, 2018-04-11:
The `name` value in your library.properties contains the disallowed character `/`. Please do the following:
1. Change the name to meet the requirements of the [Arduino Library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format).
2. Create a release or tag (make sure it matches the value of the `version` field in your library.properties).
3. Comment here so we know we can proceed with adding your library to the Library Manager index.

stechio, 2018-04-11:
1. Sorry for my oversight, now the name is fixed.
2. Sorry for my oversight, now the tag should be ok (see below).

github: https://github.com/stechio/arduino-ad-mux-lib
release: https://github.com/stechio/arduino-ad-mux-lib/releases/tag/1.0.0

Thanks

cmaglie, 2018-04-16:
done

closed 2018-04-16 by cmaglie