arduino/Arduino #7464: [Library Manager] Please add these libraries

lanselambor, 2018-04-13:
https://github.com/Seeed-Studio/Music_Shield
https://github.com/Seeed-Studio/Mutichannel_Gas_Sensor
https://github.com/Seeed-Studio/NFC_Tag_M24LR6E
https://github.com/Seeed-Studio/OLED_Display_128X64
https://github.com/Seeed-Studio/RFBee
https://github.com/Seeed-Studio/RFID_Library
https://github.com/Seeed-Studio/RTC_DS1307
https://github.com/Seeed-Studio/Seeed_QTouch
https://github.com/Seeed-Studio/SeeedMotorShieldV2
https://github.com/Seeed-Studio/Seeeduino_GPRS
https://github.com/Seeed-Studio/Small_ePaper_Shield
https://github.com/Seeed-Studio/TFT_Touch_Shield_V1
https://github.com/Seeed-Studio/TFT_Touch_Shield_V2
https://github.com/Seeed-Studio/Touch_Screen_Driver
https://github.com/Seeed-Studio/Wio_LTE_Arduino_Library

Thanks!

per1234, 2018-04-13:
https://github.com/Seeed-Studio/RFBee is a sketch, not a library, so I have struck that URL out.

lanselambor, 2018-04-13:
Thank you, https://github.com/Seeed-Studio/RFBee has been corrected with your pull request.

lanselambor, 2018-04-17:
Thank you has fixed that: https://github.com/Seeed-Studio/HDC1000

per1234, 2018-04-18:
@cmaglie I think there was some confusion regarding [RFBee](https://github.com/Seeed-Studio/RFBee). This repository contains a [sketch](https://github.com/Seeed-Studio/RFBee/tree/master/RFBee), not a library. It does have a [library.properties file](https://github.com/Seeed-Studio/RFBee/blob/master/library.properties) so it was accepted by the indexer and can be installed via Library Manager but then you get the warnings:
```
Invalid library found in E:\electronics\arduino\libraries\RFbee_V1.1_-_Wireless_arduino_compatible_node: E:\electronics\arduino\libraries\RFbee_V1.1_-_Wireless_arduino_compatible_node
```
The [pull request](https://github.com/Seeed-Studio/RFBee/pull/4) I submitted to that repository was unrelated to the Library Manager index inclusion request. It was simply intended to make the sketch more user friendly. It did not convert it to a valid library.

I suppose if you really wanted to make that sketch installable via Library Manager you could structure the repository as a dummy library with the sketch as an example but I don't think that's an appropriate use of the Library Manager index.

cmaglie, 2018-04-23:
> @cmaglie I think there was some confusion regarding RFBee. This repository contains a sketch, not a library. It does have a library.properties file so it was accepted by the indexer and can be installed via Library Manager but then you get the warnings:

Oh I see I removed RFBee from the index (well it will be truly removed on the next clean-up round in a couple of days).