arduino/Arduino #7466: Please add my library for multithreading to index

slavaza, 2018-04-13:
[https://github.com/slavaza/Thread](https://github.com/slavaza/Thread)

per1234, 2018-04-13:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

slavaza, 2018-04-14:
Ok. Thank you. I created a pre-release "Preliminary".

slavaza, 2018-04-14:
Sorry, I'm still not very good at operating github. I hope now everything is correct.

cmaglie, 2018-04-16:
done!

closed 2018-04-16 by cmaglie

slavaza, 2018-04-16:
@cmaglie  Thank you very much.