arduino/Arduino #7468: After projewt odwnloading via USBasp no more serial downloads possible

PiotrC1, 2018-04-14:
Hi,
I have noticed following problem of IDE 1.8.5 (tested with my Arduino Nano and MightyCore ATMega 664) - after downloading project (together with bootloader) using Ctrl+Shift+U (via USBasp) no more downloads using serial port are possible. If I download (burn) bootloader only - then serial downloads are possible again.

per1234, 2018-04-14:
Upload Using Programmer (Ctrl + Shift + U) only uploads the compiled sketch. It does not upload the bootloader. Thus this is the expected behavior and does not represent a bug.

closed 2018-04-14 by per1234

PiotrC1, 2018-04-14:
Sorry, that's not this case - I have replaced whole `hardware\arduino\avr\` folder from 1.8.5 package with the one from `AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\` and the result is stiill the same. Keep on debugging.

matthijskooijman, 2018-04-15:
@PiotrC1, that's exactly what @per1234 already said before, and yes, that is intentional (since you do not need the bootloader if you have a programmer, so skipping the bootloader makes things slightly faster on startup), so we'll probably leave this closed.