arduino/Arduino #7469: Serial monitor resets Arduino only once )on first startup)

PiotrC1, 2018-04-14:
If I start Serial Monitor it resets adruino. If I start it once again nor reset occurs.

per1234, 2018-04-14:
Please provide an exact set of steps we can use to reproduce this issue.

Did you close the Serial Monitor before you started it again?

Which Arduino board are you using?

PiotrC1, 2018-04-14:
I have tested two boards:

-Nano
-ATMega 644P (MightyCore)
Problem happens only with 644 (doesn't happen with Nano). Serial monitor resets Arduino only once after sketch compilation and downloading.

facchinm, 2018-04-16:
On Serial Monitor open, we use `dtr` and `rts` signals to reboot the main processor (if the board is a UNO-like). Probably your 644 completely resets if these signals are being moved, so the workaround is to add `serial.disableRTS` and `serial.disableDTR` to `boards.txt` (see https://github.com/arduino/Arduino/blob/101677bba23b6da6d55a46b89d95be0167d47d83/arduino-core/src/processing/app/Serial.java#L68 ). Feel free to open an issue to MightyCore repo to tell them about this problem. Thanks!

closed 2018-04-16 by facchinm