arduino/Arduino #7473: error: using typedef-name 'fpos_t' after 'struct'

matendie, 2018-04-15:
This issue was never resolved here (https://github.com/arduino/Arduino/issues/5630).
Still remaining as problem.

Work around should be reverting to Arduino 1.0.6 (works)

Here are conditions under which it occurs:
Upgrading Creality CR10S motherboard (creality v2.1) with creality provided modified firmware. Believed to be compiled under this 1.0.6 version, not working under new 1.0.8

Error during compiling: error: using typedef-name 'fpos_t' after 'struct' 

Source Code given by Creality: https://twitter.com/RealSexyCyborg/status/983361545580220417

Board used: Atmel Mega 2560 (https://www.amazon.com/Upgraded-Motherboard-Mainboard-Creality-3D/dp/B079H3T983)

File is corrupted at:
appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2\avr\include\stdio.h:950:33

per1234, 2018-04-15:
This is not a problem on Arduino's end. This is not a corrupted file. This is a problem caused by avr-libc adding a typedef to stdio.h which conflicts with a name previously used in a file the took from an old version of the SdFat library. The problem was fixed in Marlin a long time ago:
https://github.com/MarlinFirmware/Marlin/issues?q=fpos_t
so the only issue is that the Marlin version you're using is outdated. Either use the latest version of Marlin or else roll back to the version of Arduino AVR Boards before the toolchain update that added the `fpos_t` typedef:

1. Tools > Board > Boards Manager
2. Wait for downloads to finish
3. Click on "Arduino AVR Boards"
4. From the dropdown version menu, select "1.6.11"
5. Click "Install"
6. Wait for installation to finish.
7. Click "Close"

closed 2018-04-15 by per1234

matendie, 2018-04-16:
Got it! thank you for explaining this to me!