arduino/Arduino #7478: [Library Manager] Please, add Petri-Net library

luisllamasbinaburo, 2018-04-17:
Please, add my library Petri-Net that implements a Petri network in Arduino.
https://github.com/luisllamasbinaburo/Arduino-PetriNet

(first time I submit a library here, please let me know if I need to change anything)

per1234, 2018-04-17:
I believe you forgot to update the `name` field in your library.properties:
https://github.com/luisllamasbinaburo/Arduino-PetriNet/blob/v1.0.0/library.properties#L1
```
name=MultiTaskLib
```
You should update the `name` value and then redo your release, otherwise the library name in the Library Manager index will be stuck at MultiTaskLib, which could be confusing and also will conflict with your [other library of that name](https://github.com/luisllamasbinaburo/Arduino-MultiTask/blob/master/library.properties#L1) if you decide to add that one to the Library Manager index as well.

Please comment here when that change has been made so we know we can proceed with adding the library to the index.

luisllamasbinaburo, 2018-04-17:
Sorry, my mistake! You're totally right.
It's solved now. Thank you!