arduino/Arduino #7484: [Library Manager] Please add ProtoCentral FDC1004 library

protocentralashwin, 2018-04-18:
Please add the library for this breakout board. Library available at:

https://github.com/Protocentral/ProtoCentral_fdc1004_breakout

Thank you !

per1234, 2018-04-18:
Although your repository currently meets the requirements for inclusion in the Library Manager index, it did not at the time of the [v1.0.1 release](https://github.com/Protocentral/ProtoCentral_fdc1004_breakout/tree/v1.0.1). Please create an up to date tag or release and then comment here so we know we can proceed with adding your library.

protocentralashwin, 2018-04-19:
Thank you @per1234 , updated with new tag here:

https://github.com/Protocentral/ProtoCentral_fdc1004_breakout/releases/tag/v1.0.2