arduino/Arduino #7491: Menubar Not Working

Kimurain, 2018-04-20:
So I am rather new to the Arduino software, and my starter kit is arriving in the mail tomorrow, and I thought I would prepare a little code for a pattern I wanted to do and just wire it up tomorrow, but as I booted up the software, The menubar up top will not open a menu box, every single option just doesn't work. It just appears as if it was pressed but nothing appears. I've tried uninstalling and re-installing and nothing happens. It'll boot up like normal, but when I go to press File, Tools, or any other option. It appears as if I clicked it but no options show up. I don't know if this is human error, if my machine isn't up to par. (I'm using a little old desktop computer that runs windows 7) or what. Any help is appreciated, and I really hope that this is something fixable.

per1234, 2018-04-20:
Which version of the Arduino IDE are you using?

Kimurain, 2018-04-20:
I am using Arduino 1.8.5. The newest one I just picked off the website.