arduino/Arduino #7498: avrdude goes 100% cpu after unsuccessful uploading and closing arduino IDE

adamitsch, 2018-04-21:
When I close (interrupt Arduino IDE after unsuccessful uploading (when it is trying to connect via serial port), avrdude process consumes 100% cpu.

facchinm, 2018-04-23:
Hi @adamitsch ,
which OS / IDE version are you running? Starting from 1.8.0 (if I remember correctly) all "zombies" upload tools get killed after two minutes of inactivity. The 100% cpu behaviour is only due to avrdude getting stuck in an infinite loop. 
This should be probably tracked under https://github.com/arduino/Arduino/issues/5928

hoonto, 2018-10-26:
I saw this problem this morning. Last night I had an unsuccessful upload to board due to serial port permissions. I fixed the serial port permissions with 

```
sudo usermod -a -G dialout mmullens
sudo chmod a+rw /dev/ttyUSB0
sudo chmod a+rw /dev/ttyACM0
```

My laptop runs a lot of stuff, so didn't notice the activity until this morning at which point I saw 100% cpu utilization:

```top - 10:01:11 up 3 days, 21:14,  1 user,  load average: 2.44, 2.41, 2.27
Tasks: 372 total,   4 running, 302 sleeping,   0 stopped,   1 zombie
%Cpu(s):  9.3 us, 18.4 sy,  0.0 ni, 72.3 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem : 65920704 total, 47128480 free,  9402004 used,  9390220 buff/cache
KiB Swap:        0 total,        0 free,        0 used. 55327280 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                                                                                          
30554 mmullens  20   0   16972   5664   2220 R 100.0  0.0 779:09.58 avrdude                                                                                                                                          
30856 mmullens  20   0   16972   5696   2244 R 100.0  0.0 777:14.57 avrdude                                                                                                                                          ```

```
$ cat /etc/lsb-release
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=19
DISTRIB_CODENAME=tara
DISTRIB_DESCRIPTION="Linux Mint 19 Tara"
```

Arduino IDE 1.8.6

Hope that helps!

gibeautc, 2018-11-08:
seeing the same issue here.  Arduino IDE is not running anymore (hasnt been in a day or two) 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND     
14307 chadg     20   0   16604    680    680 R    100.0     0.0        10148:43 avrdude

closed 2018-11-23 by facchinm

EinarJohansson, 2020-12-26:
Still having this issue on Mac osx High Sierra.
Arduino version: 1.8.13

All3xJ, 2021-12-12:
[image]
still having this issue too, under ArchLinux

facchinm, 2021-12-13:
Hi @All3xJ ,
probably this code path is not being executed on close https://github.com/facchinm/Arduino/commit/bc22fd36e994b11faf5a457dade256c371265caf#diff-599f5cbf0662d5498451c870eeca912dfda5a7afd8907d66e62be228f1b11612R965 .
How are you closing the IDE?