arduino/Arduino #7499: [Library Manager] request to add CStringBuilder library

jandrassy, 2018-04-21:
location https://github.com/jandrassy/CStringBuilder

CStringBuilder is a simple class to print content to a c-string.
Builds a c-string with Print class methods into provided buffer.

jandrassy, 2018-04-21:
@per1234 Per, thank you. I merged the corrections

njh, 2020-04-09:
@jandrassy the https://github.com/jandrassy/CStringBuilder repository has been deleted - any background about why?

jandrassy, 2020-04-09:
CStringBuilder is in my StreamLib for two years now. There it has fixes for new cores.

I followed https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library and deleted the release tags two weeks ago to remove the library from index.