arduino/Arduino #7505: 1.8.5 / high CPU load

ghollf, 2018-04-23:
IDE has permanently high CPU load after startup. procexp shows for javaw permanently executing
listSerialsj.dll!sp_get_lib_version_string+0x74f
  SETUPAPI.dll!SetupDiGetClassDevsA+0x3d
    SETUPAPI.dll!SetupDiGetClassDevsExW

facchinm, 2018-04-23:
Hi @ghollf ,
this seems related with the JNI wrapper over `liblistSerials`. Are you able to see the serial ports list? Does it get populated?
`sp_get_lib_version_string` API doesn't get explicitly called by the IDE, it's quite strange it appears in your stacktrace.

ghollf, 2018-04-23:
Yes, Tools>Port lists 15 ports (mostly from USB devices like mobiles or 3G modems).

facchinm, 2018-04-23:
Probably the high cpu usage is cause by the huge number of devices (and thus syscalls to retrieve the information).
When https://github.com/libusb/libusb/issues/86 will be completed we'll be able to switch to an event-based serial listing.

ghollf, 2018-04-23:
Updated initial post. Also ports are listed immediately. Nearly all ports are associated to disconnected devices. Running `listSerialC` doesn't show any strange behaviour.

trogper, 2018-07-14:
I have found a way to reproduce the bug:
quickly toggle focus to the main IDE window, the cpu usage goes high,
give focus to serial monitor and the usage goes down again

AlexGuo1998, 2018-09-17:
Hi, @facchinm 
It seems that this issue is caused by unconnected (or disconnected) serial ports.
For example (v1,8,5, on Windows 10), plug in your Arduino board and open `serial monitor`, then unplug the board. Wait a moment, and you'll notice a high CPU load.

facchinm, 2018-09-17:
@AlexGuo1998 I've been able to reproduce it, thanks! I'll let you know as soon as there's a patch available to be tested.

trogper, 2018-09-17:
@AlexGuo1998, @facchinm
For me, the high CPU usage does occur when no board is connected and also when it still is connected.
I have tried to profile the IDE with visualVM, but have only found out it's BatikCleaner Thread

facchinm, 2018-10-23:
The serial monitor issue gets solved by #8046 , @trogper 's problem looks more like the java u171 issue (https://github.com/arduino/Arduino/issues/8009)

darkyp, 2018-12-07:
Writing just to let you know...
I have been using occasionally 1.6.6 for playing with an ESP8266. Decided to get to 1.8.8. Both experience high CPU usage spikes. With 1.6.6 they are lower. With 1.8.8 higher. services.exe followed by a javaw.exe. Have not researched in the stack trace but I'm sure it is pretty much as the one from the oridinal poster. Both versions are unzipped (though I'm an admin). Had three COM ports - one com0com pair (COM2-COM4) and an on-motherboard one COM1. Removing the com0com pair reduced the spikes. Disabled COM1 - reduced the spikes further. 
Otherwise all the ports were properly populated in the Tools -> Port menu. Now, with removed/disabled ports, naturally the Port menu is disabled. 
I do not know what is the point in having the IDE constantly enumerating the serial devices. Yes, pluging in a USB Serial while the IDE is running and making it visible might be a "user friendly" feature but this refresh might also be non-user frindly by a simple sub-menu item that reads "Refresh" or another that is checkable such as "Auto refresh".

My specs: Windows XP Pro SP3, Intel Pentium 4 CPU @ 3.00GHz, 2GB RAM. 
As I can see you telling me how much old my PC is, please don't.  Windows was installed in 2003 and is still running after several HW upgrades. I'm still very happy with it. Chrome for web browsing works.
Thanks Google and MS for dropping the support of Win/Chrome - no new updates - no issues.
Eclipse for some serious Android development is also working pretty well. Software is not meant to use the whole hardware just because it can. A good software should do its best to require the least resources.

trogper, 2018-12-07:
@darkyp how about your java runtime version? Have you tried updating it?

darkyp, 2018-12-07:
The JRE is the one that's in the 1.8.8 distribution ZIP - 1.8.0_191-b12.
1.6.6 is running with 1.8.0_60-b27.
The others that I have installed are 1.5.0_04, 1.5.0_10, 1.6.0_25, 1.7.0_25-b17.

facchinm, 2019-07-18:
Should be solved by #8046, please reopen if the issue still occurs.

closed 2019-07-18 by facchinm