arduino/Arduino #7507: [Library Manager] Please add Timer

toxnico, 2018-04-23:
Hi, I wrote a library to simply manage recurring events.

Here it is: [https://github.com/toxnico/Timer](https://github.com/toxnico/Timer)

Please tell me if the name is too generic, I can rename it if it's the case.

Thanks,

Nico

cmaglie, 2018-04-23:
> Please tell me if the name is too generic, I can rename it if it's the case.

yes, please!

what bothers me is that an include like `timer.h` is going to name-clash with a lot of core-related things.

toxnico, 2018-04-23:
@cmaglie of course, you're completely right! I didn't even think about this...
So I renamed everything to DMTimer and the repository is right here:
[https://github.com/toxnico/DMTimer](https://github.com/toxnico/DMTimer)

Thanks,
Nico

cmaglie, 2018-04-23:
:+1: 

The library has been added.

closed 2018-04-23 by cmaglie