arduino/Arduino #7513: Please update Neo7Segment Display library to v1.0.1

UnexpectedMaker, 2018-04-25:
Can you please update to Arduino library manager to point to Neo7Segment 1.0.1 (latest master)
github: https://github.com/UnexpectedMaker/Neo7Segment

Thanks very much!
Seon

per1234, 2018-04-25:
Once your library has been added to the Library Manager index, any release/tag you create that meets the Library Manager requirements will automatically be picked up by the indexer job and added. This job runs hourly so you don't need to wait long. In fact, your release has already been picked up!

So you don't need to open an issue for every release. Everything will happen automagically.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

closed 2018-04-25 by per1234

UnexpectedMaker, 2018-04-25:
Oh, I didn’t realise, thanks so much!!!!!

Seon Rozenblum | Director & Lead Code Monkey | 3 Sprockets 
www.3sprockets.com <http://www.3sprockets.com/>  |  www.linkedin.com/in/seonr/ <http://aboutme.com/seonrozenblum>  |  www.about.me/seonrozenblum

“making cool stuff for desktop, console and mobile"