arduino/Arduino #7516: Serial Monitor not working on Linux

HankB, 2018-04-25:
When opening the Serial Monitor I get 
`Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J
	at jssc.SerialNativeInterface.openPort(Native Method)
	at jssc.SerialPort.openPort(SerialPort.java:158)
	at processing.app.Serial.<init>(Serial.java:136)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$4.<init>(SerialMonitor.java:101)
	at processing.app.SerialMonitor.open(SerialMonitor.java:101)
	at processing.app.Editor.handleSerial(Editor.java:2392)
	at processing.app.Editor.lambda$buildToolsMenu$0(Editor.java:736)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

Arduino version 1.8.5
Linux version - Debian 9 (Stretch) 64 bit.

I can upload sketches without difficulty.
Serial comm between host host and Arduino Nano works using `cat` and `echo` on /dev/ttyUSB0.

matthijskooijman, 2018-04-25:
I just tried the arduino 1.8.5 64-bit linux zip download ( https://www.arduino.cc/download_handler.php?f=/arduino-1.8.5-linux64.tar.xz ) on a Stretch system, and did not get that error. Since the error talks about not resolving some native method, I suspect it is related to some native .so  library for jssc not being found, or not the right version.

I tried running arduino under `strace -f`, which shows that it access a jssc library stored in `~/.jssc`. Since I never installed that, I suspected it was autocreated. Removing the directory confirms this. Among other stuff, the strace shows this:

[pid 15973] stat("/home/matthijs/.jssc", 0x7f6e613ea8b0) = -1 ENOENT (No such file or directory)
[pid 15973] mkdir("/home/matthijs/.jssc", 0777) = 0
[pid 15973] mkdir("/home/matthijs/.jssc/linux", 0777) = 0
[pid 15973] stat("/home/matthijs/arduino-1.8.5/lib/jssc-2.8.0-arduino1.jar", {st_mode=S_IFREG|0644, st_size=154476, ...}) = 0
[pid 15973] openat(AT_FDCWD, "/home/matthijs/.jssc/linux/libjSSC-2.8_x86_64.so", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
[pid 15973] stat("/home/matthijs/.jssc/linux/libjSSC-2.8_x86_64.so", {st_mode=S_IFREG|0644, st_size=14464, ...}) = 0
[pid 15973] lstat("/home/matthijs/.jssc", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
[pid 15973] lstat("/home/matthijs/.jssc/linux", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
[pid 15973] lstat("/home/matthijs/.jssc/linux/libjSSC-2.8_x86_64.so", {st_mode=S_IFREG|0644, st_size=14464, ...}) = 0
[pid 15973] openat(AT_FDCWD, "/home/matthijs/.jssc/linux/libjSSC-2.8_x86_64.so", O_RDONLY) = 3
[pid 15973] openat(AT_FDCWD, "/home/matthijs/.jssc/linux/libjSSC-2.8_x86_64.so", O_RDONLY|O_CLOEXEC) = 3

IOW, it creates the directory and .so file inside (probably extracting
it from the jssc jar) and then opens it.

So perhaps there is a problem with your .jssc directory, or perhaps
there is an incorrect or old version in there (that is missing some
symbols?). If you *have* a directory there, please move it out of the
way rather than completely removing it, so we can investigate the root
cause of this problem.

When I remove my `~/.jssc` directory and replace it with a regular file
(effectively preventing a directory from being created there), I get the
same error:

	Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J

This suggests that Arduino can somehow now create or fill the .jssc
directory. Perhaps you ran Arduino as root before and now the .jssc
directory is owned by root?

In any case, the Arduino IDE should probably show a better error message when creating `~/.jssc` fails (provided JSSC gives some feedback about this).

HankB, 2018-04-25:
Yes, that was the issue. If ~/.jssc does not exist then `arduino` creates and populates it. In my case it was a symbolic link to a missing directory on another filesystem. With ~/.jssc renamed, running `arduino` creates it and populates it with 
```
.jssc/linux/libjSSC-2.8_x86_64.so
```
And the Serial Monitor now works. It also works with the directory moved to the other filesystem and linked. Apparently it was the broken symlink that caused the problem.

Thanks!

(I'm closing this as my original problem is resolved. You can reopen it or open a new issue if you wish to pursue a better error message.)

closed 2018-04-25 by HankB

matthijskooijman, 2018-04-26:
Thanks for the quick and detailed followup, good to hear it is solved. Good call to close this issue, I've opened a new issue for improving the error message.