arduino/Arduino #7517: Failure to create ~/.jssc directory is not properly reported

matthijskooijman, 2018-04-26:
For the serial monitor, the JSSC library is used. On linux (and possibly other OS's as well), this requires a small native library, which is automatically unpacked from the .jar file into the `~/.jssc` directory. However, when this unpacking fails, not message is shown about that. Instead, an exception is thrown and logged when opening the serial monitor:

Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J

This can be reproduced by removing `~/.jssc` and replacing it with a regular file or broken symlink. Removing permissions on the directory (which might happen when somebody runs the IDE as root) will probably also trigger it.

It would be better if the actual problem would be reported, so the user can act on that, instead of just showing the linkage error which is just a secondary symptom.

This is a followup issue for #7516, which was about the linkage error shown above (it also has a more complete backtrace).

HartOfWave, 2018-08-24:
I think I'm having this issue, there was no `~/.jssc` created when I installed the IDE. I tried creating a symlink to `/usr/share/java/jssc-2.8.0.jar` but still got the same error, so maybe it's something else. I did notice that everything in `/usr/share/java/` is owned by root, I haven'y tried charging the permissions of those files. Either way this error isn't very useful.

```
Exception in thread "Thread-42" java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J
	at jssc.SerialNativeInterface.openPort(Native Method)
	at processing.app.SerialPortList.getUnixBasedPortNames(SerialPortList.java:339)
	at processing.app.SerialPortList.getPortNames(SerialPortList.java:300)
	at processing.app.SerialPortList.getPortNames(SerialPortList.java:182)
	at processing.app.Serial.list(Serial.java:277)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:139)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2168)
	at java.lang.Thread.run(Thread.java:748)
```

cmaglie, 2018-08-24:
jssc outputs native libraries in .jssc, on my system I have:

```
$ ls -lR ~/.jssc
.jssc:
total 4
drwxr-xr-x 2 cmaglie cmaglie 4096 feb  1  2014 linux

.jssc/linux:
total 48
-rw-r--r-- 1 cmaglie cmaglie 14464 gen 23  2014 libjSSC-2.6_x86_64.so
-rw-r--r-- 1 cmaglie cmaglie 14464 gen 23  2014 libjSSC-2.7_x86_64.so
-rw-r--r-- 1 cmaglie cmaglie 14464 feb  1  2014 libjSSC-2.8_x86_64.so
```

I think you should just remove `.jssc` from your home and let jssc repopulate it.

HartOfWave, 2018-08-24:
ah thanks for that, it seems like it wasn't installing from my repo correctly (running OpenSUSE), I removed everything and installed form the tar and it's good now