arduino/Arduino #7525: [Library Manager] Please add the LINE-Notify library

TridentTD, 2018-04-28:
Hello

Here is a LINE-Notify library for ESP8266 & ESP32.
Can send message , LINE's sticker or picture to LINE application.

https://github.com/TridentTD/TridentTD_LineNotify/releases

Best regards
TridentTD.

cmaglie, 2018-05-02:
I've added the library to the index but there are some problems:

```
JOB 1528 - 2018/05/02 11:02:25 Scraping https://github.com/TridentTD/TridentTD_LineNotify.git
JOB 1528 - 2018/05/02 11:02:28 Checking out tag: 2.0
JOB 1528 - 2018/05/02 11:02:28 Error syncing library: Error generating library from repo: 'sentence' and 'url' fields must be defined
JOB 1528 - 2018/05/02 11:02:28 Checking out tag: 2.1
JOB 1528 - 2018/05/02 11:02:28 Error syncing library: Error generating library from repo: 'sentence' and 'url' fields must be defined
```
To fix this:
1) add the `sentence` and `url` fields to the library.properties
2) commit and push
3) re-do the release tag

Since the library has been added to the index, once done the above the library will be automatically published within an hour, no need to comment further here.

closed 2018-05-02 by cmaglie

per1234, 2018-05-02:
@cmaglie the library does define a `sentence` field:
https://github.com/TridentTD/TridentTD_LineNotify/blob/2.1/library.properties#L5

It also has an empty `url` field, I'm not sure whether that's allowed or not.

I notice that when I do a manual installation of the library and then check its Library Manager entry:
[image]
there is a strange handling of the [`author` field](https://github.com/TridentTD/TridentTD_LineNotify/blob/2.1/library.properties#L3):
```
author=TridentTD <fb: www.facebook.com/miniNodeMCU>
```
I wonder if the `author` or `maintainer` values are causing the indexer to not recognize the `sentence` field (and possibly `url` also)?

I notice the same error was encountered in https://github.com/arduino/Arduino/issues/7509#issuecomment-385921429, even though again the sentence field was defined. In that case it may have been caused by the forgotten closing angle brackets on author or maintainer fields
https://github.com/patou01/HC-SR04/blob/V0.2/library.properties#L3-L5
```
author=Patrick Bobbink <pbobbink@gmail.com
maintainer=Patrick Bobbink <pbobbink@gmail.com
sentence=A library to use an HC-SR04 ultrasonic sensor
```