arduino/Arduino #7527: Mac doesn't open Arduino

Edizgoldeli, 2018-04-28:
macOS High Sierra 10.13.4 doesn't open 32-bit Arduino application. I install latest Java which Version 8 update 171 (build 1.8.0_171-b11)

(Note: It says "Unable to load java runtime environment)

per1234, 2018-04-29:
Which version of the Arduino IDE are you using?

Possible duplicate of https://github.com/arduino/Arduino/issues/7213

Similar past issues:
- https://github.com/arduino/Arduino/issues/7068
- https://github.com/arduino/Arduino/issues/2384
- https://github.com/arduino/Arduino/issues/2178

Edizgoldeli, 2018-04-29:
version 1.8.5

facchinm, 2018-05-02:
Hi @Edizgoldeli,
the hourly version of the IDE is now embedding jre 1.8.0_171.
Thanks for reporting!

gvarisco, 2018-12-11:
This should be fixed. @Edizgoldeli, I'm going ahead and mark this ticket as 'closed', but feel free to re-open it in case you encounter further problems with the newer, stable version released.

closed 2018-12-11 by gvarisco

Edizgoldeli, 2018-12-11:
After many tries I realise that my usb converter is fake (I bought it from aliexpress) so I think there isn’t any problem. Sorry for this mistake. Best regards.

iPhone’umdan gönderildi