arduino/Arduino #7530: Tab names would be improved

lmihalkovic, 2018-04-29:
:) a picture is worth a 1000 words, right?!

[image]

per1234, 2018-04-30:
Perhaps, but a few words in addition to a picture could make your intention clear and also allow the issue to be found in a search.

Are you requesting that when the tabs using a full filename for label text exceed the width of the screen that the tab label text be truncated to allow more tabs to fit?

lmihalkovic, 2018-04-30:
Does it not strike u a ‘difficult to use’ as it is?

lmihalkovic, 2018-04-30:
Eg solution could be
 * smarter heuristic (compute common parts and try to find difference)
 * associate a tooltip that is visible while hovering over the tab