arduino/Arduino #7531: Can't open Arduino Lesson files

xDarkMoon, 2018-04-29:
While I was starting lesson 4, I noticed that you had to select a lesson file. I looked through the Arduino files and didn't see any. I installed Arduino with all files and don't know was wrong. 

All help will be appreciated :)

per1234, 2018-05-01:
If you provide more detail this issue would be more likely to have a resolution.
>While I was starting lesson 4

Lesson 4 of what?
>I noticed that you had to select a lesson file.

Please post the complete and exact text of the instructions referring to this "lesson file".

per1234, 2018-07-26:
Closing due to lack of response. @xDarkMoon if you ever get the time to provide the requested information I'll be happy to reopen this.

closed 2018-07-26 by per1234