arduino/Arduino #7532: Install warnings for IDE 1.8.5 on Raspbian (stretch)

cad908, 2018-04-30:
I installed the IDE onto my Raspberry Pi 3B+ from the downloaded install package:
arduino-1.8.5-linuxarm.tar.xz

I verified the downloaded file by using the sha512 hash given on the arduino.cc download page.

The unit is running Raspbian (stretch) Release 9.4
Kernel is Linux raspberry pi 4.14.34-v7 #1110

on running ./install.sh, I received the following messages:
```
Adding desktop shortcut, menu item and file associations for Arduino IDE...
touch: cannot touch '/root/.local/share/applications/mimeapps.list': No such file or directory
/usr/bin/xdg-mime: 803: /usr/bin/xdg-mime: cannot create /root/.local/share/applications/mimeapps.list.new: Directory nonexistent
done!
```
In the end, the IDE seemed to have installed properly, and I was able to compile and load a test sketch to my arduino UNO.

facchinm, 2018-05-02:
Hi @cad908 ,
this looks like a problem with Raspbian's `xdg-install` or, maybe, you are trying to install the IDE system-wide using `root` user instead than `sudo`, so the script tries to use `/root/.local/share/applications/mimeapps.list.new` instead than `/home/youruser/.local/share/applications/mimeapps.list.new` (which surely exists).
Closing as "not a bug", feel free to comment if I said somehting wrong :slightly_smiling_face:

closed 2018-05-02 by facchinm

cad908, 2018-05-02:
Hi @facchinm - Thanks much for taking a look...

I installed from the default "pi" user.

Honestly, though, I can't remember whether I prefixed "./install.sh" with sudo when I ran the install script, and now I can't find the specific instructions I used to check one way or the other... Perhaps it wouldn't have worked at all otherwise... Although I think I remember reading in another thread that escalation wasn't required.

Best,
-Chris.

NoGoCode, 2018-06-19:
I'm having the similar problem as issue #7532, with the only difference being that nothing is being installed. Not sure if I should have posted under a new issue or start a new one.

I have a very basic knowledge of what I'm doing. I've posted the details below, trying to get this running on a new PI3b+ board. I was able to successfully install this on a PI3b that was upgraded from Wheezy to Stretch and had a prior version also installed. Also, sorry for my incorrect usage of of the markup language in Github.

- Raspberrry PI: PI3b+ (Fresh install of the OS)
- ran updates which were successful
- Had a naming issue of the system, but have fixed the name and the 'install.sh' no longer throws an error about the name.
- I've verified that `RESOURCE_NAME=arduino-arduinoide` is correct in 'install.sh'
- I'm running from the default 'pi' user account with a password that has been changed.

Here is the output: sudo ./install.sh
`pi@PI4:~/Downloads/arduino-1.8.5 $ sudo ./install.sh
Adding desktop shortcut, menu item and file associations for Arduino IDE...touch: cannot touch '/root/.local/share/applications/mimeapps.list': No such file or directory
/usr/bin/xdg-mime: 803: /usr/bin/xdg-mime: cannot create /root/.local/share/applications/mimeapps.list.new: Directory nonexistent
 done!
pi@PI4:~ $ arduino
bash: arduino: command not found
pi@PI4:~ $
`

Here is the version information:
`pi@PI4:~/Downloads/arduino-1.8.5 $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

Any suggestions would be helpful.