arduino/Arduino #7533: Cant open project

hardie2611, 2018-04-30:
When I click on a project it comes up with you are not authorized to access this page.

The project is under security projects and is called online attendance system (without Ethernet)

Any help would be much appreciated

per1234, 2018-04-30:
This issue is regarding an Arduino Project Hub project:
https://create.arduino.cc/projecthub/gadget-programmers/online-attendance-system-without-ethernet-c07682
That URL redirects me to the last Project Hub I viewed and displays the "You are not authorized to access this page." message on that page:
[image]

mastrolinux, 2018-05-03:
Hi @per1234 That project is not public, how did you get the URL?

per1234, 2018-05-03:
@mastrolinux I typed the project title into Arduino Project Hub's search box:
https://create.arduino.cc/projecthub/search?q=online+attendance+system+%28without+Ethernet%29
[image]

per1234, 2018-05-07:
@mastrolinux I'm amazed that a system would be designed to display private projects in public search results. That seems like a bug to me but maybe it's hackster.io's bug rather than Arduino's.

blarralde, 2018-05-08:
Hey all, that was a bug indeed, the private project is now removed from the search.

PRAhit, 2019-02-22:
Dear @blarralde, 

There is one project on Arduino Project HUB named as Breakout Pong
http://email.mg.hackster.io/c/eJxNkN1uhCAQhZ9G7yT8uZYLL0x27WOYAUelK2AQu-nbF1ubNCGB-c7MmWHGVtJR0tK2nDJFOasZFzcmCCNKCvVoHqLraM9qVReSupksYJ57wkhsKJeWKkAh1NTAxCcFuoZJCq21EVJrOvFybZeUtr0QXcH7fExESEggjof1gRiT2RbDB5q0HDoH3a8yvKM_77Mk-BTDWqVXqCKu8LVX4RNjlRasrM-jeExVUzMm3grRH8kNBtwGdvaFuHt8DZf_XvDbj5r90Kcs1hdxONrDZYAO7HrBPRzRYIZ_P754fp2pl-ngwWEZW-sN6GjNM29pPl2ICa5M7f_-39Nxek8 and it is showing that you are not authorised to access this page. Please have a look at the provided link.

blarralde, 2019-02-22:
It seems that this project was briefly made public which is why it was in the email, but then was made private again.