arduino/Arduino #7542: Incompatibility from Arduino IDE to OneDrive

Enderbaer2308, 2018-05-01:
Hey guys,

After a few weeks of not using the Arduino IDE, I had the following problem:
I tried to compile my sketch and received the following errormessage:

```
readlink C:\Users\USER\OneDrive\Documents\Arduino\libraries\Adafruit_PWMServoDriver\examples\pwmtest\pwmtest.ino: The system cannot find the file specified.

Error compiling for board Arduino/Genuino Uno.

```
A bit of research on the internet later I found out, that there is an incompatibility between the Ardunio IDE and OneDrive. Is there any chance, that this will be fixed in the future?

Thanks for your help

Enderbaer

Enderbaer2308, 2018-05-01:
My results were so far to move the "Arduino" folder from my OneDrive directory to a simple harddrive. That worked, but I'd like to use my files also on other devices.

matthijskooijman, 2018-05-01:
Closing as duplicate of arduino/arduino-builder#254.

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2018-05-01 by matthijskooijman