arduino/Arduino #7546: The file "twi.c" uses "true" and "false", even though it's C and none of the headers define true or false.

mortie, 2018-05-02:
The file [hardware/arduino/avr/libraries/Wire/src/utility/twi.c](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/Wire/src/utility/twi.c) doesn't seem to compile, because `true` and `false` isn't defined.

matthijskooijman, 2018-05-02:
Can you tell us a bit more about your setup? This file seems to compile fine for most users in the Arduino IDE, where AFAICT it is also compiled as C, not C++. Looking at the file itself, it includes "Arduino.h", which I think defines true/false (either directly, or perhaps by including stdbool.h).

mortie, 2018-05-03:
You're right, Arduino.h includes `stdbool.h`. I was using the unofficial Arduino.h file for ATTiny cores, this issue should probably have been opened there, and not here. Sorry.

closed 2018-05-03 by mortie