arduino/Arduino #7566: не делает капиляцию

sokol36, 2018-05-05:
Arduino: 1.8.0 (Windows 10), Плата:"Arduino Nano, ATmega328"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\sokol\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10800 -build-path C:\Users\sokol\AppData\Local\Temp\arduino_build_766332 -warnings=more -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Мои документы\моделизм\ардуино\nRF24L01-master\nRF24L01 tests\nrf_listen_air\nrf_listen_air.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\sokol\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10800 -build-path C:\Users\sokol\AppData\Local\Temp\arduino_build_766332 -warnings=more -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Мои документы\моделизм\ардуино\nRF24L01-master\nRF24L01 tests\nrf_listen_air\nrf_listen_air.ino
Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "C:\Users\sokol\AppData\Local\Temp\arduino_build_766332\sketch\nrf_listen_air.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\sokol\AppData\Local\Temp\arduino_build_766332\sketch\nrf_listen_air.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\sokol\AppData\Local\Temp\arduino_build_766332\sketch\nrf_listen_air.ino.cpp" -o "C:\Users\sokol\AppData\Local\Temp\arduino_build_766332\preproc\ctags_target_for_gcc_minus_e.cpp"
E:\РњРѕРё РґРѕРєСѓРјРµРЅС‚С‹\РјРѕРґРµР»РёР·Рј\Р°СЂРґСѓРёРЅРѕ\nRF24L01-master\nRF24L01 tests\nrf_listen_air\nrf_listen_air.ino:2:22: fatal error: nRF24L01.h: No such file or directory

compilation terminated.

Используем библиотеку SPI версии 1.0 из папки: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
exit status 1
Ошибка компиляции для платы Arduino Nano.

per1234, 2018-05-05:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2018-05-05 by per1234

sokol36, 2018-05-05:
огромное спасибо вы помогли моей проблеме .я добавил библиотеку и все за работало и еще раз большое спасибо