arduino/Arduino #7568: Arduino CLI verify returns stat RunningChromeVersion: no such file and doesn't work

pnuzhdin, 2018-05-05:
Run `/Applications/Arduino.app/Contents/MacOS/Arduino --verify  test.ino`.
Got:
```
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
stat /Users/pnuzhdin/Library/Application Support/Google/Chrome/RunningChromeVersion: no such file or directory
```
Btw, `/Applications/Arduino.app/Contents/MacOS/Arduino --get-pref` works fine.

facchinm, 2018-05-07:
It looks like `--verify` is spilling outside its "normal" folders (maybe due to a symlink)?
Could you try executing 
```
SKETCHLOCATION=$PWD
cd /Applications/Arduino.app/Contents/MacOS/
./Arduino --verify $SKETCHLOCATION/test.ino
```
and see if it also happens?
Also, please enable "Verbose output during compilation" in under Preferences and paste here all the output so we can understand why it's trying to access a nonsense directory.

pnuzhdin, 2018-05-07:
Without verbose:
```
pnuzhdin@grayportable.Dlink:/Applications/Arduino.app/Contents/MacOS$ ./Arduino --verify $SKETCHLOCATION/test.ino
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
stat /Users/pnuzhdin/Library/Application Support/Google/Chrome/RunningChromeVersion: no such file or directory
```
With verbose:
```
pnuzhdin@grayportable.Dlink:/Applications/Arduino.app/Contents/MacOS$ ./Arduino -v --verify $SKETCHLOCATION/test.ino
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/pnuzhdin/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/pnuzhdin/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/pnuzhdin/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=230400 -ide-version=10805 -build-path /var/folders/5j/twh1h1xj4s3db4f8p0g2r5zh0000gn/T/arduino_build_623991 -warnings=default -build-cache /var/folders/5j/twh1h1xj4s3db4f8p0g2r5zh0000gn/T/arduino_cache_457401 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/pnuzhdin/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/pnuzhdin/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/pnuzhdin/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -verbose /Users/pnuzhdin/test.ino
stat /Users/pnuzhdin/Library/Application Support/Google/Chrome/RunningChromeVersion: no such file or directory
```

Still the same error. Also, if I call `stat` for the same strange path it actually works:
```
pnuzhdin@grayportable.Dlink:/Applications/Arduino.app/Contents/MacOS$ stat '/Users/pnuzhdin/Library/Application Support/Google/Chrome/RunningChromeVersion'
16777220 8608910505 lrwxr-xr-x 1 pnuzhdin staff 0 13 "May  5 22:03:19 2018" "May  5 22:03:19 2018" "May  5 22:03:19 2018" "May  5 22:03:19 2018" 4194304 0 0 /Users/pnuzhdin/Library/Application Support/Google/Chrome/RunningChromeVersion
```

facchinm, 2018-05-07:
Ok, the problems seems related with `test.ino` not being inside a `test` folder. If you try to open it with double click, the IDE should prompt you to create a folder and move the file there.
The Arduino specification does only allow a sketch if it's contained in a folder with the same name; probably the builder doesn't enforce this behaviour enough and tries to scan every subdirectory.
I'm moving the issue to https://github.com/arduino/arduino-builder .

closed 2018-05-07 by facchinm