arduino/Arduino #7569: "Export compiled Binary" with .eep file

Dragon-Knight, 2018-05-06:
When exporting a binary file, only a hex file is created. When both in the temp folder there are both hex and eep files.
Please, add the ability to export both program data and eeprom data.

ubidefeo, 2018-05-23:
If I may add, it would be great if this option could create a "build" folder in the sketch folder containing everything including the `.elf` file.
Some users (like me) may want to do debugging and have to go scavenge for the `.elf` in the temp folder which is always erased and changes name on compile, hence the path will change

Dragon-Knight, 2018-05-23:
I fully support. The .elf file is also very often used and it is very inconvenient to look for it in the temp folder.

ubidefeo, 2018-06-04:
@facchinm , could you also consider my extension as an improvement and add a build folder with all that's needed to the sketch folder? :)

stevecolt, 2018-07-23:
Count me in on this request, a separate build directory under sketch folder would be ideal
Steve

alexceltare2, 2019-06-26:
This will be greatly useful. An option in Arduino to choose a path where to export the .bin/.hex/.eep upon compilation will help to quickly distribute a binary file for production. I fully support this idea.

ubidefeo, 2019-06-26:
hi there, fellows :)
has any of you tried Arduino-CLI?
when building it also produces an .elf file local to the sketch.
for users comfortable with the command line it's really neat.

My workflow is based on Sublime Text 3 and a simple build script to compile/upload