arduino/Arduino #7570: [Library] RGB565 Color Definitions List

Rudey0107, 2018-05-06:
Please tell me if I uploaded my library to GitHub properly.

https://github.com/Rudey0107/RGB565Colors/tree/master

per1234, 2018-05-06:
There are several things you'll need to do in order to be eligible for inclusion in the Library Manager index:

1. Rename the file [RGB565Colors](https://github.com/Rudey0107/RGB565Colors/blob/master/RGB565Colors) to RGB565Colors.h
2. Add a file named library.proerties to the root of your library, following this specification: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
3. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2018-07-26:
Closing due to inactivity. @Rudey0107 if you ever get around to making your library compliant with the requirements listed in my last reply you are welcome to comment here and I'll reopen the issue.

closed 2018-07-26 by per1234