arduino/Arduino #7573: Proxy configuration not saved

matthjes, 2018-05-07:
Hi,

I'm working behind a corporate proxy, so I need to configure the proxy in the IDE. However, when I enter the proxy credentials, close the configuration window, and reopen it again, the credentials are missing. This also happens when I enter the credentials directly into the preferences.txt: as soon as I restart the IDE the credentials get erased.

This happens with version 1.8.5 and 1.9.0 beta.

This problem is also described in this thread: https://forum.arduino.cc/index.php?topic=392432.0

facchinm, 2018-06-28:
Hi @matthjes ,
it looks quite strange; a possible explanation is that you don't have write access to your user-wide `preferences.txt` (which lives in Arduino15 folder, the exact location depends on the operating system in use, `C:\Users\username\AppData\Local\Arduino15` in case you are in Windows).
Just to make sure, if you change any other property (like the sketchbook location or the verbosity) are they getting saved on restart?

per1234, 2018-09-26:
@matthjes are you still having this problem? We're still waiting for you to provide the information requested by facchinm.

per1234, 2018-11-03:
I can reproduce this problem with Arduino IDE 1.8.7 hourly build 2018/10/24 04:33 and Arduino 1.9.0-beta build 87 using Windows 10.

1. Delete or rename your preferences.txt file to simulate a fresh installation. You can find the location of preferences.txt by clicking the link at the line following [**File > Preferences > More preferences can be edited directly in the file**.
2. File > Preferences
3. Click on the "Network" tab.
4. Select "Manual Proxy Configuration"
5. Enter something in all the input fields.
6. Click "OK"
7. File > Preferences
8. Click on the "Network" tab. The **Manual Proxy Configuration** Username and Password fields are empty.
9. Click "OK"
10. File > Quit
11. Open preferences.txt. The `proxy.manual.password` and `proxy.manual.username` properties are blank.
12. Start the Arduino IDE.
13. File > Preferences
14. Click on the "Network" tab.
15. Select "Auto-detect proxy settings".
16. Enter something in all fields.
17. Click "OK"
18. File > Preferences
19. Click on the "Network" tab. The **Auto-detect proxy settings** Username and Password fields are still filled.
20. Click "OK"
21. File > Quit
22. Open preferences.txt. The `proxy.manual.password` and `proxy.manual.username` properties are set to the values you entered in the **Auto-detect proxy settings** Username and Password fields.
23. Start the Arduino IDE.
24. File > Preferences
25. Click on the "Network" tab.
26. Select "Manual Proxy Configuration"
27. Click "OK"
28. File > Preferences
29. Click on the "Network" tab. The **Manual Proxy Configuration** Username and Password fields are now filled with the values you entered in the Auto-detect proxy settings Username and Password fields.

So there is clearly a bug in the **Manual Proxy Configuration** system. There is no problem writing any other preference and even `proxy.manual.hostname` and `proxy.manual.port` work as expected.

It's also not clear whether it was intentional for **Auto-detect proxy settings** and  **Manual Proxy Configuration** to share the same `proxy.manual.username` and `proxy.manual.password` properties or that's a bug.

solarjoe, 2019-01-24:
I can confirm this behaviour on Win10, IDE 1.8.8
Auto-config writes and reads to the `manual` fieldes in `preferences.txt`.

Also behind a corporate firewal wall and must use a proxy with authentification, Tried several combinations in GUI and `preferences.txt` without success.

Does the auto configuration also look for the environment variables `HTTP_PROXY` or `HTTPS_PROXY` on Windows?

solarjoe, 2019-01-25:
I can download the `http://downloads.arduino.cc/packages/package_index.json` from my browser and I tried to configure the Arduino IDE the same way, without luck.

When trying to update the boards list I get 

`Caused by: java.io.IOException: Unable to tunnel through proxy. Proxy returns "HTTP/1.1 407 Proxy Authentication Required"`

I noticed in another thread (https://github.com/arduino/Arduino/issues/5887) that it could also show something like 

```
Using proxy HTTP @ /192.168.0.254:3128
Caused by: java.io.IOException: Received invalid http status code from server: 416
```

This is the section from my `perferences.txt`:
```
proxy.manual.hostname=x.x.x.x
proxy.manual.password=xxxx
proxy.manual.port=8080
proxy.manual.type=HTTP
proxy.manual.username=xxxx
proxy.pac.url=
proxy.type=manual
```

solarjoe, 2019-01-25:
Could it be related to this?

https://stackoverflow.com/questions/41505219/unable-to-tunnel-through-proxy-proxy-returns-http-1-1-407-via-https

facchinm, 2019-01-25:
PR created here https://github.com/arduino/Arduino/pull/8436 , binaries should be ready in an hour more or less.

solarjoe, 2019-01-25:
I just tested this and it works when I add

`-Djdk.http.auth.tunneling.disabledSchemes=""`

to the file `arduino.l4j.ini` or `arduino_deubg.l4j.ini` where you can pass additional parameters, see  
http://launch4j.sourceforge.net/docs.html#Additional_jvm_options

solarjoe, 2019-01-25:
I was not testing the new binaries, just modified the installation I had (1.8.8).

facchinm, 2019-01-25:
If you could test the PR binaries it would be better since every OS has its own different launcher and adding flags is a bit tricky in some of them (OSX for example)

ThomasBe, 2019-02-01:
> I just tested this and it works when I add
> 
> `-Djdk.http.auth.tunneling.disabledSchemes=""`
> 
> to the file `arduino.l4j.ini` or `arduino_deubg.l4j.ini` where you can pass additional parameters, see
> http://launch4j.sourceforge.net/docs.html#Additional_jvm_options

I had the same issue and until 1.8.7 everything was fine. With 1.8.8. I try a lot to get my proxy settings working.

By just adding the mentioned line into  `arduino_deubg.l4j.ini` everything is fine with the same settings like with 1.8.7.

solarjoe, 2019-02-01:
@ThomasBe, do you also see the issue with the proxy setting not being saved when you enter them in the UI?

ThomasBe, 2019-02-01:
@solarjoe , I only use the "auto_proxy_part" and everything works regarding modyfing in GUI or via preferences.txt.

JorinL, 2019-02-04:
With PR #8436 I have the following experience:

Filling the manual proxy config with "abc" clicking ok and opening it again shows only hostname and port filled.
looking into the preferences.txt shows:

proxy.manual.hostname=
proxy.manual.password=
proxy.manual.port=
proxy.manual.type=HTTP
proxy.manual.username=
proxy.pac.url=
proxy.type=none

First thought caused by my job - I should not be able to fill in "abc" into a pure number field :-D
Event with a number in it only hostname and port is saved in the gui

Filling the auto proxy config with "abc" clicking ok and opening it again shows everything filled.
looking into the preferences.txt shows:

proxy.manual.hostname=
proxy.manual.password=
proxy.manual.port=
proxy.manual.type=HTTP
proxy.manual.username=
proxy.pac.url=
proxy.type=none

No Proxy shows me nothing special in the config but stays selected in the gui even after reopening

Thomseeen, 2019-04-04:
Beeing on Unbuntu 18.04 with Arduino 1.8.9 I managed to make the IDE remember my proxy credentials by editing the `preferences.txt` but i still get the `"HTTP/1.0 407 Proxy Authentication Required"` error. 

> I just tested this and it works when I add
> 
> `-Djdk.http.auth.tunneling.disabledSchemes=""`
> 
> to the file `arduino.l4j.ini` or `arduino_deubg.l4j.ini` where you can pass additional parameters, see
> http://launch4j.sourceforge.net/docs.html#Additional_jvm_options

Can someone help me find those `.ini`-files to add the needed launch parameters to my 1.8.9 installation?

solarjoe, 2019-04-04:
Should be in the same folder as your Arduino IDE binary.

Thomseeen, 2019-04-05:
On my Win10-machine, yes, under Ubuntu, sadly not...

facchinm, 2019-09-18:
@solarjoe @JorinL @ThomasBe @Thomseeen @matthjes could you give a spin to latest IDE 1.8.10 and report if the proxies are still not applied?
https://github.com/arduino/Arduino/pull/9023 is included and was meant to solve most of the downloads issues.

ThomasBe, 2019-09-19:
> @solarjoe @JorinL @ThomasBe @Thomseeen @matthjes could you give a spin to latest IDE 1.8.10 and report if the proxies are still not applied?
> #9023 is included and was meant to solve most of the downloads issues.

I checked 'my problems' with the latest version (1.8.10) and internet connection stuff works fine for me without any additional work around. 

So I just entered my network settings using the automatic configuration via configuration url.

facchinm, 2019-09-19:
@TomasRoj thanks a lot for testing. I'm closing this as fixed, if it's not please comment here and we'll reopen it.

closed 2019-09-19 by facchinm

QGB, 2022-09-24:
```
正在下载 107-Arduino-APDS-9950@1.0.0
Failed to install library: 107-Arduino-APDS-9950:1.0.0.
Error: 13 INTERNAL: 无法下载库: Get "https://downloads.arduino.cc/libraries/github.com/107-systems/107_Arduino_APDS_9950-1.0.0.zip": proxyconnect tcp: dial tcp :0: connectex: The requested address is not valid in its context.

```

per1234, 2022-09-24:
Hi @QGB. Arduino IDE 2.x uses a completely different preferences store from Arduino IDE 1.x. The `preferences.txt` file is completely ignored by Arduino IDE 2.x, so it is expected that the proxy configuration from that file has no effect on Arduino IDE 2.x.

You can configure the proxy of Arduino IDE 2.x by doing the following:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Click the "**Network**" tab.
1. Select the "**Manual proxy configuration**" radio button.
1. Configure proxy as needed for your network.
1. Click the <kbd>**OK**</kbd> button.

If you have any questions or problems, please post over on the Arduino Forum. I'll be glad to help you out over there:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

QGB, 2022-09-24:
I delete all `C:\Users\qgb\AppData\Local\Arduino15` and still not work .

I have to manually download and save to cache file 

`C:\Users\qgb\AppData\Local\Arduino15\staging\packages\esp32-2.0.5.zip`