arduino/Arduino #7576: New library - SDS dust sensors

lewapek, 2018-05-07:
Hi,
I've created library for Nova Fitness SDS dust sensors.
https://github.com/lewapek/sds-dust-sensors-arduino-library
The library attempts to provide easy-to-use abstraction over Laser Dust Sensor Control Protocol V1.3
I tried to write some docs and examples to make facilitate the use of the library.
I'd be glad if you add this lib to Libraries Manager ;)

Regards
lewapek

per1234, 2018-05-07:
Parentheses are not permitted in library.properties `name` field for libraries in the Library Manager index.

Please change your `name` value to something compliant with the [Arduino Library Specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format) and then update the tag/release. Once that is complete, comment here so we know we can proceed to add your library to the index.

lewapek, 2018-05-07:
I changed the name field and updated tag/release. It should be ok now

cmaglie, 2018-05-10:
Done!

closed 2018-05-10 by cmaglie