arduino/Arduino #7582: 1.8.5 crashing on macOS 10.13.3

derekkinsman, 2018-05-09:
When opening any files/projects from File → Examples, File → Sketchbook, File → Open Recent, File → Open, and File → New the application crashes. Upon reopening the newly opened/created file will be open.

facchinm, 2018-05-22:
Hi @derekkinsman ,
I asked an OSX user with 10.13.3 to test it and she couldn't reproduce the bug.
Can you launch the Arduino executable via commandline and paste here the error log (probably a stacktrace)? 
Thanks

derekkinsman, 2018-05-25:
@facchinm I seem to be unable to reproduce the problem. I suspect that the problem came from either a library or a tool. The only thing that has changed since filing the bug is a few (community & official) libraries have updated.

Closing as the problem seems to have sorted itself out. ¯\_(ツ)_/¯

unknowndomain, 2018-06-17:
I have found this issue extremely reproducible. I work at a University and whenever working with students I tend to work on their laptop, I've found as many as 10 different machines including my own where it crashes opening the blink example, including my own machine, there doesn't appear to be a common set of libraries, it just keeps happening on 1.8.5.

facchinm, 2018-06-18:
@unknowndomain I'd love if you could provide a stacktrace of what is happening (simply launch Arduino executable from a shell ( as `/Applications/Arduino.app/Contents/MacOS/Arduino` for example). Are you (or your students) running OSX in a non-english language? OSX 10.13 has been particularly cumbersome to support (due to bugs like [this](https://github.com/arduino/Arduino/issues/6548))

derekkinsman, 2018-06-22:
@facchinm here's my stacktrace. Steps to reproduce:
1. Open Arduino
2. Open Blink example
3. Open another example (I specifically opened Bare Minimum)

```
united-network-for-law-and-enforcement:~ derekkinsman$ /Applications/Arduino.app/Contents/MacOS/Arduino
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0x00007fff2ff21efa, pid=97970, tid=0x0000000000000307
#
# JRE version: Java(TM) SE Runtime Environment (8.0_144-b01) (build 1.8.0_144-b01)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.144-b01 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [CoreFoundation+0x15defa]  _CFRelease+0x51a
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/derekkinsman/hs_err_pid97970.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Abort trap: 6
united-network-for-law-and-enforcement:~ derekkinsman$
```

facchinm, 2018-06-28:
Hi @derekkinsman ,
it looks like the problem related with Java itself (not really something we can fix, except for updating the bundled jre).
It would be nice if you could test the IDE hourly build on your Mac since we updated the bundled jre to 1.8.0_171. Thanks!

derekkinsman, 2018-06-28:
@facchinm I can't reproduce the error on `Arduino 1.8.6 Hourly Build 2018/06/18 03:12`

closed 2018-07-09 by facchinm

per1234, 2018-11-04:
@technoblogy you need to use the [hourly build](https://www.arduino.cc/en/Main/Software#hourly) for testing. Otherwise the problem you are reporting might have already been fixed.

technoblogy, 2018-11-04:
It's intermittent so difficult to reproduce. I'd been using 1.8.7 for a couple of days before it happened. Please reopen this issue so other users can report if they have this problem.

per1234, 2018-11-04:
As you well know, anyone is free to comment here whether the issue is open or closed. But we need people to test with the hourly build of the Arduino IDE. This was suspected to be a Java issue and the hourly build uses a newer Java version than Arduino IDE 1.8.7 does.

technoblogy, 2018-11-04:
OK, thanks, I'm using the hourly build now.