arduino/Arduino #7584: New library: PxMatrix LED matrix library for ESP8266 and ESP32

2dom, 2018-05-10:
Could you please add https://github.com/2dom/PxMatrix

Thanks!

cmaglie, 2018-05-18:
done

2dom, 2018-05-18:
Great ... thanks!

2dom, 2018-07-06:
I have continued developing the library and posted two new versions on github...how do they find their way into the library manager?

2dom, 2018-07-06:
I have continued developing the library but the new tagged releases do not appear in the library manager ... what am I doing wrong?

per1234, 2018-07-06:
I believe the reason is that you changed the `name` value in library.properties from "PxMatrix LED MATRIX for Adafruit GFX" to "PxMatrix LED MATRIX library":
https://github.com/2dom/PxMatrix/commit/d1a5a2cd15627ff61920e9a705d4480a10d871a8

If that's the new name you want the library indexed under then cmaglie can update the entry.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-change-my-librarys-name

2dom, 2018-07-06:
Yes ... I would like to change the name. Would you mind updating the entry? Thanks!

cmaglie, 2018-07-14:
done

closed 2018-07-14 by cmaglie