arduino/Arduino #7591: Add repository URL field to the Library Manager index

per1234, 2018-05-12:
The Library Manager indexer job must work off a list of repository URLs but this information is not shared in the index file:
http://downloads.arduino.cc/libraries/library_index.json
The repository URL is not needed for the Arduino IDE's Library Manager but it could be very useful for alternative uses of the index by 3rd parties.

`url` uses the library `name` as defined by library.properties rather than the repository name and there is no guarantee they are the same. `website` does not necessarily point to the repository.

scls19fr, 2018-05-13:
This could help both #7567 and #6646

https://github.com/scls19fr/arduino_libraries_search can help to automate a part of the task of retrieving a repository url from archive url and website url but sometimes that's not enough.

njh, 2018-05-21:
This would be really useful.

For the [arduinolibraries.info](https://www.arduinolibraries.info) website I am currently having to track the repository URLs manually myself in a Google spreadsheet (where they can't be guessed).

scls19fr, 2018-05-21:
It could be great if you could merge your manually tracked data (when repositories url exists and is still available)

cmaglie, 2018-05-21:
The repository URL is now available on the library_index.json:

```
    {
      "name": "WiFi101",
      "version": "0.10.0",
      "author": "Arduino",
      "maintainer": "Arduino \u003cinfo@arduino.cc\u003e",
      "sentence": "Network driver for ATMEL WINC1500 module (used on Arduino/Genuino Wifi Shield 101 and MKR1000 boards)",
      "paragraph": "This library implements a network driver for devices based on the ATMEL WINC1500 wifi module",
      "website": "http://www.arduino.cc/en/Reference/WiFi101",
      "category": "Communication",
      "architectures": [
        "*"
      ],
      "types": [
        "Arduino"
      ],
      "repository": "https://github.com/arduino-libraries/WiFi101.git",      <---------
      "provides_includes": [
        "WiFi101.h"
      ],
      "url": "http://downloads.arduino.cc/libraries/github.com/arduino-libraries/WiFi101-0.10.0.zip",
      "archiveFileName": "WiFi101-0.10.0.zip",
      "size": 209803,
      "checksum": "SHA-256:649f69cb2833d591c8962e99b4b5b85c89d1fc58e6e09e4b2966f9010191e120"
    },
```

closed 2018-05-21 by cmaglie

njh, 2018-05-24:
This is fantastic, thanks @cmaglie.

Having done some checks with the new `repository` field, I have found that some repositories have moved.

Should I create a ticket with a list of updates?

scls19fr, 2018-05-31:
I noticed in https://github.com/bblanchon/ArduinoJson/commit/7e4fcb0868c9da02f49283ea06dadeeee498765d
that `repository` field is (was) also used by some libraries in `library.properties`

Maybe [wiki](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata) should document such a field usage.

cmaglie, 2018-05-31:
> that repository field is (was) also used by some libraries in library.properties

It's not part of the `library.properties` because it's not needed (the library indexer has his own list).

Moreover, if at some point the repository is moved the old releases will retain the old `repository` value (because the git history can not be changed).

@njh 
> Should I create a ticket with a list of updates?

Github automatically redirects to the new url, so it's not really a problem right now.
It should be implemented as a check inside the indexer to update the URL automatically, I'll keep a note about that, but low priority.

scls19fr, 2018-06-01:
just a few words to say that Particle https://docs.particle.io/guide/tools-and-features/libraries/#library-properties-fields uses both license and repository field.

scls19fr, 2018-06-01:
ok indexer have it's own list but how can you tell indexer that you are using https://spdx.org/licenses/GPL-3.0-or-later.html and not https://spdx.org/licenses/GPL-3.0.html

LICENSE files are the same but SPDX short identifier are differents

per1234, 2018-06-01:
@scls19fr this issue is regarding adding the repository URL to the Library Manager index, which cmaglie has already been kind enough to do for us. Discussion of specifying the license in the Library Manager index is off topic for this issue and you already have [another issue](https://github.com/arduino/Arduino/issues/6646) for that.

orkungedik, 2024-04-25:
Hi,

Even though I am able to see my library in https://downloads.arduino.cc/libraries/library_index.json, why I can't see on Arduino IDE library list?

Thank you for your support.

Orkun Gedik

per1234, 2024-04-25:
@orkungedik I can see it:

[image]

So if you can't see it then it means the problem is specific to your system. This repository is only used for tracking bug reports and feature request, not for user support. The appropriate place for support requests is Arduino Forum:

https://forum.arduino.cc/

If you post about your problem over there, I'll be happy to provide assistance.