arduino/Arduino #7595: eeprom tutorials missing code

sterretje, 2018-05-13:
Not sure what the correct section is to post this.

I've checked eeprom tutorials, https://www.arduino.cc/en/Tutorial/EEPROMCrc and https://www.arduino.cc/en/Tutorial/EEPROMClear. There might be more.

Both show the below message where the code is supposed to be

> SORRY, There is an error at our code repository, please inform to web@arduino.cc

per1234, 2018-05-15:
I went through all the library example pages and found these have the above error:

- https://www.arduino.cc/en/Tutorial/SoftwareSerialExample
- https://www.arduino.cc/en/Tutorial/TwoPortReceive
- https://www.arduino.cc/en/Tutorial/BarometricPressureSensor
- https://www.arduino.cc/en/Tutorial/DigitalPotControl
- https://www.arduino.cc/en/Tutorial/DigitalPotentiometer
- https://www.arduino.cc/en/Tutorial/MasterReader
- https://www.arduino.cc/en/Tutorial/MasterWriter
- https://www.arduino.cc/en/Tutorial/SFRRangerReader
- https://www.arduino.cc/en/Tutorial/EEPROMClear
- https://www.arduino.cc/en/Tutorial/EEPROMRead
- https://www.arduino.cc/en/Tutorial/EEPROMWrite
- https://www.arduino.cc/en/Tutorial/EEPROMCrc
- https://www.arduino.cc/en/Tutorial/EEPROMGet
- https://www.arduino.cc/en/Tutorial/EEPROMIteration
- https://www.arduino.cc/en/Tutorial/EEPROMPut
- https://www.arduino.cc/en/Tutorial/EEPROMUpdate

A related previous issue (though that one has been mostly resolved):
https://github.com/arduino/Arduino/issues/5794

SimonePDA, 2018-05-17:
The libraries have been moved to a new path, therefore the code link was broken. Now the list provided by per1234 is fixed. 

Thanks

closed 2018-05-17 by SimonePDA