arduino/Arduino #7596: Add datetime release field to the Library Manager index

scls19fr, 2018-05-13:
Hello,

something might be very useful...
Date/Time of a version release might be something very interesting to know.
I noticed that this information is missing when working on https://github.com/scls19fr/arduino_libraries_search

I have been able to count number of version for each library but wanted to know when it was release first, if it was released recently...

It can be a good indicator to know if a library is still maintained or not.

Related issue #7591 from @per1234 and #7567

Kind regards

cmaglie, 2018-05-18:
We do not save this information. As suggested by @per1234 we could add the repository url to the library index so you can retrieve it by scraping from github?

cmaglie, 2018-05-22:
The repository url is now available.

closed 2018-05-22 by cmaglie

scls19fr, 2018-05-22:
For future release of library I think it will be better to keep a track of date/time
Repository url can help to get date/time of a release only if release are correctly tagged (which may not be the case for each library)

scls19fr, 2018-05-25:
@cmaglie could you reopen this?
I really think that it's safer to keep a track of date/time of release on Arduino infrastructure side (ie not to rely too much on repositories that can be renamed, deleted, badly tagged...)

cmaglie, 2018-05-28:
Libraries releases are actually based on repositories tags: each release has a corresponding tag and viceversa (if the developer removes a tag the corresponding library release is removed after a while). IMHO on the library-indexer side is much simpler if the release time is taken directly from the git timestamp.

> I really think that it's safer to keep a track of date/time of releases on Arduino infrastructure side (ie not to rely too much on repositories that can be renamed, deleted, badly tagged...)

The only "unsafe" case I can see is if the library developer intentionally touch the timestamp of a git tag... but is this really relevant?
I mean, even if we keep the timestamp server-side, it can always be faked because an "evil" developer may:

1) remove a release
2) wait until it disappear from the index
3) re-release it