arduino/Arduino #7598: java.io.IOException: Cannot run program ERROR

fnguyen99, 2018-05-14:
Hi,
We spent 4 hours yesterday trying to solve this problem, and couldn't find anything on the web.
Can you please provide useful suggestion to sharper1999@gmail.com?

Thank you.
Fred

================
Arduino: 1.8.6 Hourly Build 2018/05/10 01:33 (Windows 10), Board: "Arduino/Genuino Uno"

Sketch uses 2492 bytes (7%) of program storage space. Maximum is 32256 bytes.
Global variables use 208 bytes (10%) of dynamic memory, leaving 1840 bytes for local variables. Maximum is 2048 bytes.
.
java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2197)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

fnguyen99, 2018-05-14:
> In those 4 hours, did you try downloading the 1.8.5 (stable release) ZIP
> file and run that version? Does 1.8.5 work, or also fail?
>
> You might also try temporarily disabling any anti-virus software, and
> Windows Defender. Sometimes these cause trouble.

Who is paying for my 4 hours of wasted time on version 1.8.5? ...

I thought 1.8.5 was stable? That's the one that is released on your website
https://www.arduino.cc/en/Main/Software

Are you telling us, you didn't test this version before releasing to users?
I am using Windows 10.

I also tried the beta version, and the same error occurred.

There is no anti virus software interfering, and if it did wouldn't it give
some type of warning to allow running?

Please advise what these errors mean, and how to work around it?

-- 

=====================
straight from ...
the Sharpshooter
sharper1999@gmail.com
=====================

Arduino: 1.8.6 Hourly Build 2018/05/10 01:33 (Windows 10), Board: "Arduino/Genuino Uno"

Sketch uses 2492 bytes (7%) of program storage space. Maximum is 32256 bytes.
Global variables use 208 bytes (10%) of dynamic memory, leaving 1840 bytes for local variables. Maximum is 2048 bytes.

java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2197)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

facchinm, 2018-05-15:
Found the bug, a fix is ongoing. Thanks for reporting, but always use stable builds (1.8.5 in this case) if you need something stable.

dcetin01, 2018-12-11:
I too, recently had a similar 'error' and have always used 1.8.5 build, please help!!
Here is a copy of the error:

Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

Sketch uses 4420 bytes (13%) of program storage space. Maximum is 32256 bytes.
Global variables use 483 bytes (23%) of dynamic memory, leaving 1565 bytes for local variables. Maximum is 2048 bytes.
java.io.IOException: Cannot run program "C:\Users\dcetin01\Arduino\hardware\tools\avr/bin/avrdude": CreateProcess error=193, %1 is not a valid Win32 application
 at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
 at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
 at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
 at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
 at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
 at processing.app.SketchController.upload(SketchController.java:713)
 at processing.app.SketchController.exportApplet(SketchController.java:686)
 at processing.app.Editor$DefaultExportHandler.run(Editor.java:2168)
 at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=193, %1 is not a valid Win32 application
 at java.lang.ProcessImpl.create(Native Method)
 at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
 at java.lang.ProcessImpl.start(ProcessImpl.java:137)
 at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
 ... 8 more
An error occurred while uploading the sketch
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_115740: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_115740
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_269568: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_269568
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_305492: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_305492
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_545055: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_545055
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_696625: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_696625
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_721834: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_721834
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_795191: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_795191
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_920364: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_920364
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_930976: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_930976
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_96981: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_96981
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\U8glib: C:\Users\dcetin01\Documents\Arduino\libraries\U8glib
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_115740: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_115740
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_269568: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_269568
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_305492: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_305492
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_545055: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_545055
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_696625: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_696625
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_721834: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_721834
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_795191: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_795191
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_920364: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_920364
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_930976: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_930976
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\arduino_96981: C:\Users\dcetin01\Documents\Arduino\libraries\arduino_96981
Invalid library found in C:\Users\dcetin01\Documents\Arduino\libraries\U8glib: C:\Users\dcetin01\Documents\Arduino\libraries\U8glib

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.﻿

facchinm, 2018-12-11:
@dcetin01 your bug is different as seems related with an antivirus activity. Try whitelisting `"C:\Users\dcetin01\Arduino\hardware\tools\avr/bin/avrdude"` (maybe you'll have to reinstall the IDE, it depends on your antivirus) and you should be fine.