arduino/Arduino #7599: Problem with A4 Arduino M0

warriorfenix26, 2018-05-15:
Hi i have a problem with measures of A4 input in arduino M0.

I have this code to measure the voltage in A3 and A4. A3 is connect with Digital pin 3, and A4 is connect with Digital pin 4.

Code:

`  analogReadResolution(12);
  delay(10);  
  SerialUSB.println("SALIDA LOW");
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);  
  digitalWrite(3, LOW);
  SerialUSB.print("A3: ");
  SerialUSB.println(analogRead(A3)*(3.3 / 4096.0));  
  digitalWrite(4, LOW);
  SerialUSB.print("A4: ");
  SerialUSB.println(analogRead(A4)*(3.3 / 4096.0));  
  delay(2000);
  SerialUSB.println("SALIDA HIGH");  
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);  
  digitalWrite(3, HIGH);
  SerialUSB.print("A3: ");
  SerialUSB.println(analogRead(A3)*(3.3 / 4096.0));  
  digitalWrite(4, HIGH);
  SerialUSB.print("A4: ");
  SerialUSB.println(analogRead(A4)*(3.3 / 4096.0));    
  delay(2000);`

Result:

SALIDA LOW
A3: 0.01
A4: 1.08
SALIDA HIGH
A3: 3.08
A4: 1.23
SALIDA LOW
A3: 0.02
A4: 1.28
SALIDA HIGH
A3: 3.10
A4: 1.22
SALIDA LOW
A3: 0.02
A4: 1.21
SALIDA HIGH
A3: 3.10
A4: 1.24

A4 pin ever have a 1.4v.

Please help!

facchinm, 2018-05-16:
Hi @warriorfenix26, would you mind moving the issue to https://github.com/arduino/ArduinoCore-samd ? The issue could be related with M0 pinmuxing, so A4 is not really the one labeled as such. I'm checking it, thanks

warriorfenix26, 2018-07-10:
thanks, it works!

closed 2018-07-10 by warriorfenix26