arduino/Arduino #7600: Broken web pages on arduino.cc

gratefulfrog, 2018-05-15:
I was reading the pages: 

- https://www.arduino.cc/en/Tutorial/DigitalPotControl
- https://www.arduino.cc/en/Tutorial/BarometricPressureSensor

And the embedded Code link fails with this message:

"SORRY, There is an error at our code repository, please inform to web@arduino.cc"

The address: web@arduino.cc  also fails !

I hope this helps!

Cheers,
Bob

per1234, 2018-05-15:
Thanks so much for reporting this!

I'm going to merge the reports of the error message where code should be with the related issue https://github.com/arduino/Arduino/issues/7595.

If you discover this error on any pages other than those listed at https://github.com/arduino/Arduino/issues/7595#issuecomment-389136400 and https://github.com/arduino/Arduino/issues/5794#issuecomment-272541873 please let us know.

Let's make this issue solely about the invalid email address provided in that error message, which I have confirmed.

https://www.arduino.cc/en/Main/ContactUs gives the email address `webmaster (at) arduino.cc` so maybe the email address should be changed to that.

I actually think it would be better to just direct people to report the issue to the tracker via https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md. I don't like reporting bugs to an email address because I have no way of knowing whether a hundred people have already done the same before me and I'm just wasting my time.

mastrolinux, 2018-05-16:
web@arduino.cc now works again and goes to the support team. They will take care of it in the future.

per1234, 2020-10-05:
Since the email address is reported to be fixed, I'll close this as solved.

closed 2020-10-05 by per1234