arduino/Arduino #7608: Update to the Pinduino library: https://github.com/elyons/pinduino

elyons, 2018-05-18:
Hi, 

I have updated the pinduino library (already in the Library Manager).  The latest version is available: https://github.com/elyons/pinduino

Some new arduino users have contacted me asking for this update.

Thanks,
-Eric Lyons

elyons, 2018-05-18:
I realize that the library.properties file should get queried to alert your systems of the updates, but that doesn't appear to be happening.  Any advice on how to fix that on my side?

cmaglie, 2018-05-18:
You must create a new git tag for the new release, at the moment there is only 1.0: https://github.com/elyons/pinduino/releases

Once the tag is created the new release will be published automatically on the library manager within an hour.

elyons, 2018-05-21:
Thanks for the help.  

I added the new release tag 3 days ago and it hasn't been automatically published on the library manager: https://github.com/elyons/pinduino/tags

I must be missing something -- can you advise?

cmaglie, 2018-05-21:
The problem is that the library changed name:

```
JOB 771 - 2018/05/21 15:25:51 Scraping https://github.com/elyons/pinduino.git
JOB 771 - 2018/05/21 15:25:51 Checking out tag: 1.0
JOB 771 - 2018/05/21 15:25:51 Release pinduino:1.0.0 already loaded, skipping
JOB 771 - 2018/05/21 15:25:51 Checking out tag: 1.2.6
JOB 771 - 2018/05/21 15:25:51 Release Pinduino:1.2.6 has wrong library name, should be pinduino
```

Would like to keep `Pinduino`? If yes, I need to change it in the index.

elyons, 2018-05-21:
Yes, it would be great to keep it as "Pinduino".  Thanks for identifying that problem.

cmaglie, 2018-05-25:
Done!

closed 2018-05-25 by cmaglie