arduino/Arduino #7609: Allow editing of files in subdirectories from the IDE

Pharap, 2018-05-19:
Currently the IDE does not count files within subdirectories as part of the sketch and won't open tabs for them when a sketch is opened.
(It will however include them when compiling.)

It would be very useful if code files from subdirectories were counted as part of the sketch and opened in tabs when the main `.ino` file is opened.
This would encourage better structuring of program code and would remove one of the issues that people sometimes cite as a reason for avoiding the IDE.

per1234, 2018-05-19:
I'm not trying to argue against this but I would like to point out that some projects benefit from the Arduino IDE not opening files in subdirectories. The most noteworthy is Marlin. The latest release of Marlin has 276 tabs. Most Marlin users only need access to Configuration.h and Configuration_adv.h, the files that allow them to configure the firmware. All those other tabs only make it difficult to use. Marlin 2.0.0 moves all source files to a subdirectory so when you open it with the Arduino IDE you have only 3 tabs: Marlin.ino (which only contains some comments), Configuration.h, and Configuration_adv.h.

I sometimes like to bundle library dependencies with a sketch. This allows me to send it to someone as a self-contained package so there is no need for the user to go through the extra complication of installing libraries before using the sketch. I only want to expose the simple sketch code to the sort of person I do that for. They can easily get started studying that sketch code and experimenting with modifying it. If all the library code was shoved in their face right from the start they would be very overwhelmed.

So if this is done I'd like to see some sort of UI where the files in subdirectories are not just dumped in as tabs. Instead, there should be some way to selectively expand/collapse subdirectories as needed. Maybe the subdirectory name would be presented as a tab, then when you click on that tab it adds a new tab bar row that contains the subdirectory's files/folders as tabs.

lmihalkovic, 2018-05-19:
Marlin is what i took as my ‘test lab’ for improving the ide. My goals were:
* still has to work as is today for normal users
* be able to close tabs
* fast startup time
* fast switching between tabs using keyboard
* access all the code (yes, folders included)

aside from cleaning the 1.9.x codebase, i also added a file browser as i did 2y ago. In the end, i like the result... and i can look at marlin semi-comfortably (much more than before)

For situations like marlin, it would be nice to remember whats open/closed and support reordering the tabs manually.. but i didnt care enough yet to do it

Pharap, 2018-05-19:
@per1234 That's a fair point.

Ideally it would be good to perhaps have a sort of collapsable directory tree navigator or something on the left or right, like other IDEs often have, and then if tabs are closable it would be possible to open/close files as needed, and as you say, only open the top level by default.

I have no idea how difficult that would be to pull off because I don't know what the UI is based on (e.g. AWT, Swing).

lmihalkovic, 2018-05-19:
@pharap from experience.. takes much longer the first time you do it than the second time around. 

the first version looked like: https://github.com/lmihalkovic/Arduino/blob/master/images/arduino-dark.png

but it could not close tabs. now it is the first thing i wrote

lmihalkovic, 2018-05-19:
@per1234 although your ‘meta tab’ idea is a clever trick, an implementation on the 1.8.xx or 1.9.xx (i have not looked at the changes in the last 2 weeks) codebase is not just a walk in the park. If you go down that path, you might as well do a file browser on the side (yes, u need extra code for handling the tree, but in the end that also opens other doors - like direct link to the cache for flushing, or link to the build folder itself ... blah blah blah)

lmihalkovic, 2018-05-21:
Lazy open.. and close. works even with file explorer disabled.

[image]

per1234, 2020-09-27:
I'm closing this as a duplicate of https://github.com/arduino/Arduino/issues/4821, but I'll add a link back in a comment over there to make sure this nice discussion won't be forgotten.

closed 2020-09-27 by per1234