arduino/Arduino #7612: DropboxManager library

lucasromeiro, 2018-05-20:
https://github.com/lucasromeiro/DropboxManager

per1234, 2018-05-20:
Although your repository currently meets all the requirements for addition to the Library Manager index, it did not at the time of the last release (because the library was in a subfolder of the repository). Please create (or redo) a tag or release with your library in the current form.

Once you have done this please comment here so we know we can proceed.

lucasromeiro, 2018-05-20:
Hello, now is OK!

lucasromeiro, 2018-05-21:
Hello, Something missing?

per1234, 2018-05-21:
It looks to as if everything is in order, which is why I removed the "Waiting for feedback" label.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So there is no further action required from you.

lucasromeiro, 2018-05-21:
Cool i'm learning Thank you!
How can I request an update in the library when I need it?

per1234, 2018-05-21:
You only need to update the `version` field in your library.properties file and then create a new tag or release. The Library Manager indexer job does an hourly scan of all the repositories that have been added to the index and picks up any new tags. So your new releases will automatically be added and available from Library Manager shortly after you create it without any need to request an update.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

cmaglie, 2018-05-25:
Library added, sorry for the delay.

closed 2018-05-25 by cmaglie