arduino/Arduino #7614: No error highlight after "No such file or directory" error

per1234, 2018-05-20:
Using Arduino IDE 1.9.0 Beta Build 58 or Arduino IDE 1.8.6 Hourly 2018-05-15 10:41 w/ Windows 10 64 bit

Compile the following sketch:
```cpp
#include <foobar.h>
void setup() {}
void loop() {}
```
Compilation fails:
```
C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_114310\sketch_may20d.ino:1:20: fatal error: foobar.h: No such file or directory
```
but there is no error highlight on line 1, as would be expected with a compilation error caused by sketch code.

I'm not sure how difficult this would be to resolve, but "No such file or directory" is probably the most common error we see on the Arduino forum. The users encountering this error are usually very new to Arduino (they haven't learned they need to install libraries yet) and so any additional clues to the problem could be very helpful.

facchinm, 2018-05-21:
It looks like the problem is the "fatal" word that  breaks the regexp.
Fixing it this way should at least highlight the line
```patch
diff --git a/arduino-core/src/cc/arduino/Compiler.java b/arduino-core/src/cc/arduino/Compiler.java
index 3f1ffb4e1..94a515150 100644
--- a/arduino-core/src/cc/arduino/Compiler.java
+++ b/arduino-core/src/cc/arduino/Compiler.java
@@ -133,7 +133,7 @@ public class Compiler implements MessageConsumer {
     }
   }
 
-  private static final Pattern ERROR_FORMAT = Pattern.compile("(.+\\.\\w+):(\\d+)(:\\d+)*:\\s*error:\\s*(.*)\\s*", Pattern.MULTILINE | Pattern.DOTALL);
+  private static final Pattern ERROR_FORMAT = Pattern.compile("(.+\\.\\w+):(\\d+)(:\\d+)*:\\s*(fatal)?\\s*error:\\s*(.*)\\s*", Pattern.MULTILINE | Pattern.DOTALL);
 
   private final File pathToSketch;
   private final Sketch sketch;
```

About the helper, it would need a few refactoring on that part (since now we only match exact errors).

per1234, 2018-05-21:
>About the helper, it would need a few refactoring on that part (since now we only match exact errors).

I'm not sure what you meant by that. If you're referring to what I said:

>Sometimes there will be a helpful comment near #include directives in a sketch that gives some information regarding where to find the library dependency.

by that I only meant that sometimes the author of a sketch will leave a comment in the code about the library dependency. If that line was highlighted after the "No such file or directory" error it would draw the user's attention to any comment the sketch author might have left near that line. For example:
```cpp
#include <foobar.h>  // Available from: http://github.com/per1234/foobar
```

If you're talking about some magical check of the Library Manager index for a library that matches the missing filename (without the extension), or even just a link to open a Library Manager search for it, that could be nice. Adding the library.properties `include` property to the Library Manager index and search scope (https://github.com/arduino/Arduino/issues/7604) would help this since that would make it more likely that the library would be found with a search for the filename.

cmaglie, 2018-05-21:
I changed the regexp as @facchinm suggested, now the error is correctly highlighted.

commit d4a12205 (2018-05-21): Highlight 'fatal' errors during build

Fix #7614
closed 2018-05-21 by cmaglie