arduino/Arduino #7621: Library manager exception

billytheme, 2018-05-23:
Was trying to update a library, and got a null pointer error. Error attached
[temp.txt](https://github.com/arduino/Arduino/files/2029272/temp.txt)

facchinm, 2018-05-23:
Hi @billytheme ,
thanks for sharing. Which IDE version are you using? Hourly or Beta?

per1234, 2018-09-26:
@billytheme are you still having this issue? We're still waiting for you to tell us which IDE version you're using.

billytheme, 2019-07-18:
Sorry, I totally missed this, and I no longer have the offending project. I believe the build was the beta build however. I will leave this open in case this proves helpful, and give required information as memory servers, but given the age, please feel free to close it if you consider it no longer helpful.