arduino/Arduino #7622: Clickable URLs in sketch comment block

tobozo, 2018-05-23:
[component: IDE/User Interface]
[type: improvement]

While github understands a URL shouldn't be clickable in a source code snippet, Arduino IDE seems to be at the opposite.

`sektch.ino`

```
/*
 *  Code repo example: http://github.com/arduino/Arduino/
 *
 */
```

Arduino IDE is trying (and failing on Linux) to launch a browser every time I click -intentionnaly or accidentally- on a URL in a comment block.

So my question is: how can I completely disable this option, or make it ineffective? I've tried blanking the value in the prefs, setting /dev/null and even using a "do nothing" binary to no effect.

I don't mind fiddling in the Arduino IDE code and doing this myself but I have no idea where to look for.

On the other hand it would be nice to have a checkbox or a pref flag to enable/disable this feature.

lmihalkovic, 2018-05-23:
SketchTextArea.java .. if memory serves

tobozo, 2018-05-23:
Thanks @lmihalkovic!

The amount of awaiting PR's is impressive on this project, is that worth a feature request or a PR for this?

tobozo, 2018-05-23:
```
diff --git a/app/src/cc/arduino/view/preferences/Preferences.java b/app/src/cc/arduino/view/preferences/Preferences.java
index 0f8d980..309c8cf 100644
--- a/app/src/cc/arduino/view/preferences/Preferences.java
+++ b/app/src/cc/arduino/view/preferences/Preferences.java
@@ -129,6 +129,7 @@ public class Preferences extends javax.swing.JDialog {
     arduinoNotRunningLabel = new javax.swing.JLabel();
     checkboxesContainer = new javax.swing.JPanel();
     displayLineNumbersBox = new javax.swing.JCheckBox();
+    setHyperlinksEnabledBox =  new javax.swing.JCheckBox();
     enableCodeFoldingBox = new javax.swing.JCheckBox();
     verifyUploadBox = new javax.swing.JCheckBox();
     externalEditorBox = new javax.swing.JCheckBox();
@@ -253,6 +254,9 @@ public class Preferences extends javax.swing.JDialog {
     displayLineNumbersBox.setText(tr("Display line numbers"));
     checkboxesContainer.add(displayLineNumbersBox);
 
+    setHyperlinksEnabledBox.setText("Enable Hyperlinks");
+    checkboxesContainer.add(setHyperlinksEnabledBox);
+
     enableCodeFoldingBox.setText(tr("Enable Code Folding"));
     checkboxesContainer.add(enableCodeFoldingBox);
 
@@ -727,6 +731,7 @@ public class Preferences extends javax.swing.JDialog {
   private javax.swing.JComboBox comboWarnings;
   private javax.swing.JLabel comboWarningsLabel;
   private javax.swing.JCheckBox displayLineNumbersBox;
+  private javax.swing.JCheckBox setHyperlinksEnabledBox;
   private javax.swing.JCheckBox enableCodeFoldingBox;
   private javax.swing.JButton extendedAdditionalUrlFieldWindow;
   private javax.swing.JCheckBox externalEditorBox;
@@ -826,6 +831,8 @@ public class Preferences extends javax.swing.JDialog {
 
     PreferencesData.setBoolean("editor.linenumbers", displayLineNumbersBox.isSelected());
 
+    PreferencesData.setBoolean("editor.hyperlinks_enabled", setHyperlinksEnabledBox.isSelected());
+
     PreferencesData.setBoolean("editor.code_folding", enableCodeFoldingBox.isSelected());
 
     PreferencesData.setBoolean("upload.verify", verifyUploadBox.isSelected());
@@ -900,6 +907,8 @@ public class Preferences extends javax.swing.JDialog {
 
     displayLineNumbersBox.setSelected(PreferencesData.getBoolean("editor.linenumbers"));
 
+    setHyperlinksEnabledBox.setSelected(PreferencesData.getBoolean("editor.hyperlinks_enabled"));
+
     enableCodeFoldingBox.setSelected(PreferencesData.getBoolean("editor.code_folding"));
 
     verifyUploadBox.setSelected(PreferencesData.getBoolean("upload.verify"));
diff --git a/app/src/processing/app/syntax/SketchTextArea.java b/app/src/processing/app/syntax/SketchTextArea.java
index c13d4ca..c230674 100644
--- a/app/src/processing/app/syntax/SketchTextArea.java
+++ b/app/src/processing/app/syntax/SketchTextArea.java
@@ -93,6 +93,9 @@ public class SketchTextArea extends RSyntaxTextArea {
   }
 
   private void installFeatures() throws IOException {
+
+    setHyperlinksEnabled(PreferencesData.getBoolean("editor.hyperlinks_enabled", true));
+
     setTheme(PreferencesData.get("editor.syntax_theme", "default"));
 
     setLinkGenerator(new DocLinkGenerator(pdeKeywords));
```