arduino/Arduino #7623: Support needed - Issue (PROGMEM / attribute ignored)

NRW998, 2018-05-23:
Hi GitHub Arduino community, 

Since a few weeks I have an issue with the arduino development environment. I am using an ePaper library which is provided by the manufacturer. This library worked fine till a few weeks ago. From this point I got an error message. But the point is that i didn't change anything in the library or in the source code. I am using the same compiler and the same settings. I also reinstalled everything but i got the same issue. 

The issue says that the attribute which has to be saved in the PROGMEM (internal memory of the board) will be ignored. But because of the ignoring of the attribute, my code does not work anymore. 

Maybe it depends on an Arduino Update which causes that the attribute will be ignored?

Do you have an idea about this? 

In the attachment you can find the screenshot of the error message and the library which is used (provided by the manufacturer). 

Thanks a lot in advance for your help.:) 

[image]
[Small_ePaper_Shield-master.zip](https://github.com/arduino/Arduino/files/2031364/Small_ePaper_Shield-master.zip)

matthijskooijman, 2018-05-23:
That code uses PROGMEM on class members and function arguments, neither of which are supported. I think they never were, though earlier compiler versions accepted this without warning (and might have needed them to pass PROGMEM pointers around).

Instead, you must use the PROGMEM attribute on *variables* to pass them into PROGMEM. Tracking if a pointer points into RAM or PROGMEM must be done manually, not using the PROGMEM keyword on the pointers (this is the part I think changed in the compiler, but I think that change was already in 1.5.x, maybe even 1.0.x). Maybe you recently enabled warnings in the preferences (so these warnings were hidden until recently)?

Looking at the code for that library, it actually puts non-PROGMEM stuff into these pointers (e.g. `gate_source`), so it should be fine to ignore this PROGMEM attribute.

Hence, I do not think these warnings are indeed the cause of any problems you are having.

As for debugging your problem itself - this issue tracker is not for getting help with projects, only for reporting bugs in the Arduino IDE. I suggest you ask the manufacturer for an updated library, or ask on the Arduino forums for help.

sandeepmistry, 2019-09-16:
Hi @NRW998, please see @NRW998's comment above.

closed 2019-09-16 by sandeepmistry