arduino/Arduino #7626: Serial Plotter Freezes

yesIamHasi, 2018-05-24:
Serial Plotter Freezes after x-axis is greater than 2000000

madsdyd, 2020-04-07:
Could you please provide a sketch that demonstretes this, and an estimate for the time taking to reach this state? In other words: How to reproduce?

per1234, 2021-04-05:
I have not been able to reproduce this, even using Arduino IDE 1.8.5, which would have been the current version at the time of the report. Since not enough information was provided to investigate, I'll go ahead and close this.

@yesIamHasi. If you are still able to reproduce the issue with the latest hourly build of the Arduino IDE and will comment here with all the relevant information, I'll be happy to reopen this issue.

closed 2021-04-05 by per1234