arduino/Arduino #7629: Ardunio IDE ignoring manually added board in Boards.txt

RussyCG, 2018-05-24:
I have added a board to the Boards.txt file under "C:\Program Files (x86)\Arduino\hardware\arduino\avr" and saved the file. Yet when I restart the IDE it does not appear in the Boards lists.

IDE Version: 1.8.4

Board Added:
###############################################

atmega328bb.name=AVRISP mkII w/ ATmega328
atmega328bb.upload.using=arduino:avrispmkii
atmega328bb.upload.maximum_size=30720
atmega328bb.upload.speed=57600
atmega328bb.bootloader.low_fuses=0xE2
atmega328bb.bootloader.high_fuses=0xDA
atmega328bb.bootloader.extended_fuses=0x05
atmega328bb.bootloader.file=optiboot/optiboot_atmega328.hex
atmega328bb.bootloader.unlock_bits=0x3F
atmega328bb.bootloader.lock_bits=0x0F
atmega328bb.build.mcu=atmega328p
atmega328bb.build.f_cpu=8000000L
atmega328bb.build.core=arduino:arduino
atmega328bb.build.variant=standard

###############################################

matthijskooijman, 2018-05-24:
Perhaps you updated the AVR core manually? If so, the updated version is used, which is stored in your user directory (inside a "arduino15" subdirectory IIRC).

RussyCG, 2018-05-24:
You rock. Thank you. 

I changed the boards.txt under the Ardunio15 folder, restarted IDE and like magic it was there again.

closed 2018-05-24 by per1234